Mechanical-Advantage/AdvantageKit #199: Cannot listen for changes on a LoggedSendableChooser, & other network inputs.

blaze-developer, 2025-06-22:
Sometimes, it is desirable to listen to changes on a chooser. For example:

```java
// Listen for auto changes
autoChooser_.onChange(cmd -> {
    Pose2d startingPose = cmd.getStartingPose();
    drivebase_.setPose(startingPose);
    questnav_.setPose(startingPose);
});
```

However, with a LoggedNetworkChooser, this is impossible without breaking log replay functionality, something close is like this:

```java
// Listen for auto changes
autoChooser_.getSendableChooser().onChange(name -> {
    Pose2d startingPose = autoChooser_.get().getStartingPose();
    drivebase_.setPose(startingPose);
    questnav_.setPose(startingPose);
});
```

However, the onChange event relies on non-deterministic data.

A method for these classes that listens for changes in a reproducible manner would be preferred, but I am unsure how to implement this.

jwbonner, 2025-06-22:
I think this would be relatively straightforward to implement. The `onchange` method just needs to register a callback that is triggered from the `periodic` method of the `LoggedDashboardChooser` when the value changes. The callback would be triggered synchronously from the main robot loop based on the logged input value, so it would be deterministic.

blaze-developer, 2025-06-22:
I was unaware there was a periodic loop in it but now that I know that it makes a lot of sense

closed 2025-10-17 by blaze-developer